ondrej1024/crelay #1: HIDAPI relay support?

Dh0mp5eur, 2015-03-05:
Hello,
I would like to know the USB relay based on HIDAPI are supported or hard to add support?
https://github.com/darrylb123/usbrelay

I'm interested by HTTP API.

Thx

ondrej1024, 2015-03-05:
Thanks for the link to this project. I didn't know this type of relay card with USB HIDAPI. Looks interesting, even though the manufactures web site www.dcttech.com has disapeared, so it might be difficult to get hold of these cards in the future.

I have just asked the guys over there form the usbrelay project if they want to contribute and integrate their driver software into crelay. It shouldn't be too difficult. Their code seems quite straight forward.

We'll see if they are interested,
Ondrej

Dh0mp5eur, 2015-03-05:
Thanks
This cards are a very good price on aliexpress (~10$).
And they are more simple (1 or 2 relay).
I just need a card with a single relay.
Supported cards are too expensive for my small project.

ondrej1024, 2015-03-07:
I have added support for these HID API compatible relay cards in the hidapi_experimantal branch. However I don't have this kind of card and cannot test. Do you have one and can test this? Would be great.

Dh0mp5eur, 2015-03-07:
I will receive soon.
Thx

Dh0mp5eur, 2015-03-30:
I tested and it works perfectly.
Thx

ondrej1024, 2015-03-30:
Great, thanks for testing. 
HID API support is now merged into master branch and officially released in version 0.7.  Issue resolved.
Ondrej

closed 2015-03-30 by ondrej1024