ondrej1024/crelay #3: crelay for OpenWRT

jabss, 2015-04-24:
Hello,

Could this work on OpenWRT (linux-based) routers?
It would be great to be able to control relays with the only device that is turned on 24h/day in my house :-)

Thanks,
Joaoabs

ondrej1024, 2015-04-24:
Hi,
I haven't tried this myself but don't see why it shouldn't work on a Linux router. The crelay software is designed to be independent from the hardware it runs on. As long as you have a USB port accessible and you can meet the software dependencies for the build. 
The dependencies are:
- for Sainsmart relays: libftdi library http://www.intra2net.com/en/developer/libftdi/
- for HID API relays: libhidapi: http://www.signal11.us/oss/hidapi/
- for Conrad relay: patched cp210x kernel driver http://www.silabs.com/products/mcu/pages/usbtouartbridgevcpdrivers.aspx

If you don't need the support for all relay types, you can exclude the once you don't need from the build and therefore don't have to worry about their specific dependencies.

Cheers, Ondrej

ondrej1024, 2015-05-05:
Hi Joaoabs,
did you succeed in running crelay on our OpenWRT router ?

jabss, 2015-05-05:
Hi Ondrej,

Unfortunately I couldn't, I'm unable to include the HID API and compile it
successfully...
Some guys from the forum told me that libusb could replace HID API, but I
can't get it compiled either....
Cross compiling software for OpenWRT with different libs is not easy...

I think some guys did it, I found a polish forum with some stuff, but I'm
unable to understand, and google translator didn't help that much...
(http://eko.one.pl/forum/viewtopic.php?id=11103/)

Thanks,
Joao

ondrej1024, 2015-05-05:
If you choose the Sainsmart relay, than you could do without the HID API. It can be excluded from the build.

ondrej1024, 2015-05-05:
BTW, in the OpenWRT repository source on GitHub I can see the hidapi folder with a makefile:
https://github.com/openwrt/packages/tree/master/libs/hidapi
So maybe someone has already built a package for that.

jabss, 2015-05-06:
Hi Ondrej,

I wasn't aware of that. I'll try to compile this over the weekend.

Thanks,
Joao Silva

ondrej1024, 2015-05-06:
I don't know what processor architecture your router has, but there are ready made hid-api packages in the OpenWRT package repository, e.g (for AT91).
http://downloads.openwrt.org/snapshots/trunk/at91/generic/packages/packages/

You could have a look and check if you find the package for your router there.

Ondrej

ondrej1024, 2015-07-13:
I jabss, any progress on this?

jabss, 2015-07-17:
Hi Ondrej,

Unfortunately not, I couldn't progress so I gave up on this type of relays.
My knowledge on compiling is somewhat limited. I already did some openwrt
kernel compilations in the past, but always following recipes...
So I started using other type of relays that have serial<-->USB interface.
The kind that you send codes like "0x51" to the /dev/ttyUSBx and the relay
turns on/off.

But I'm again exited that this could be included in OpenWRT development.
Maybe I can have some use to my HID relays after all. :-)

Thanks for your support,
João

thess, 2015-08-03:
Hi all -

I have created an OpenWrt package for crelay. It builds and appears to be functional on at least one platform (ar71xx). There are a few issues that should be addressed when and if possible.

A couple of additional command-line options to allow post-install configuration would help alot. Modifying a package's source code in the OpenWrt build environment is not really practical. Also, most folks install pre-built packages.
* Specify alternate http listening port (default 8000)
* GPIO pin assignments. Most OpenWrt platforms have a number of active GPIO pins controlling leds and buttons which you would want to avoid and not interfere with their operation. Perhaps default should be none?
* Specify interface/driver to use (all are included)

I also have a put together a cp210x patch to add the gpio ioctls for the OpenWrt kernel module builds (3.18 and 4.1)  which I will submit to the main OpenWrt source repo shortly. I don't know why Silicon Labs has not submitted this version to the upstream Linux kernel it seems harmless enough.

Proposed crelay package can be found:
https://github.com/thess/OpenWrt-packages/tree/crelay-import
(look in utils/crelay)

I'll stage the cp210x patches in the package contents and some directions on how to use them (a manual step is required) until they are in the main OpenWrt repo.

thess, 2015-08-08:
I have a brief follow-up on cp210x support -- Neither OpenWrt nor the mainstream linux kernel folks will accept the inclusion of ioctl support for the GPIO pins in the cp210x driver. It has been suggested that one use libusb to send control messages to the device if that is desired. I suggest that this avenue be considered in the near term. Ongoing development work is most likely to include support for GPIOs on these types of devices in the future.

ondrej1024, 2015-08-12:
@thess, thanks a lot for your work. I was on vacation and saw only now that things are progressing. :smile:

I really appreciate your contribution and comments. I hope to find some time in the next days to implement the changes you suggest.

* I will make the server listening port and IP address configurable. IP address configuration could be useful especially on a router to listen only on the LAN and not on the WAN port.
* GPIO pin assignment will be configurable, nedd to think about a good default value.
* Configuration parameters should probably be read from a .conf file, otherwise too many command line parameters will make things confusing
* Specifying the relay card driver should not be necessary, because the card is detected automatically. If no USB card is found, crelay defaults to the GPIO driver if the GPIO sysfs is found. Relay cards can also be swapped on the fly once the daemon is running.

ondrej1024, 2015-08-12:
Regarding the cp210x driver, I knew that Silabs had proposed their patch for the mainline kernel but it was rejected because their solution doesn't use standard Linux interfaces but introduces some arbitrary IO controls. Interesting to hear that there is some work being done to include support in libusb. Do you have a link with some more information on this?
If (and when) there will be an alternative to the current Silabs driver to talk to the Conrad card, I will implement support in crelay. Until then we should probably disable support for it in the crelay OpenWRT package (not include the Conrad card driver).

thess, 2015-08-13:
Looking at the libusb API, it seems to be fairly simple to send device specific control messages to read/write internal registers 
controlling GPIOs. Given the substantial number of VID/PIDs using the CP210x chips, you will probably need to probe the USB device 
similar to what you do now and try to identify the chip type and variant. (The sample driver from silabs has most of the info you 
need)

The linux-usb effort is to most likely give these devices full GPIO support in the future. The libusb route should be sufficient for 
just poking a couple of relays for now.

/ted

ondrej1024, 2015-08-13:
Interesting idea to use libusb to talk to the cp210x chip. Looks like I would need to translate the function call `usb_control_msg` from the kernel driver to `libusb_control_transfer` from libusb. There are some odd things in the driver code that are not quite clear to me but I will give it a try. Would be great to be able to control the Conrad card without the need for the Silabs driver.

ondrej1024, 2015-08-14:
It works :laughing: 
I wrote some test code and I am able to control the Conrad relay card *without* the cp210x driver. This will make things much easier now. I am getting a new version of crelay out which integrates these changes. Hope to have it ready in a couple of days.

commodo, 2015-08-17:
Hello,

So, I was following the other packaging-crelay thread ( this one : https://github.com/openwrt/packages/issues/1569 ) for a couple of weeks.

I'll just add my work here.
Maybe @thess can take a look and converge some elements from me.
Seems that @ondrej1024 cleaned up the code and my Makefile is a bit simpler than thess'
My branch is: https://github.com/commodo/packages/commits/crelay
Commit is: https://github.com/commodo/packages/commit/e43ebf2107dc69d84a6a2cd3b295d3043d983863

ondrej1024, 2015-08-21:
@thess, I just pushed the new version 0.9 with much more flexible configuration handling via crelay.conf file. There are quite a lot of changes. I also updated the documentation so you should be able to see how the configuration is done now. Let me know if you need anything else to update the package.

thess, 2015-08-21:
Great - updated OpenWrt package to V0.9 and added the conf. 
Send me a note or post a note in the OpenWrt/packages repo if you have another update.
Close this for now?

ondrej1024, 2015-08-22:
That was quick, thanks. Will inform you when a new version comes out.
Just one last question before I close this discussion. Where are the binary OpenWRT packages for crelay available?

thess, 2015-08-22:
http://downloads.openwrt.org

crelay should be in the Development Snapshots (indexed by platform) in a day or 2. There was a small build problem with hidapi which prevented crelay from building successfully. I just checked-in the fix so it will be at least 24hrs before the buildbots produce a complete set of packages for crelay.

If you can’t wait, you could always setup your own OpenWrt build environment and DIY.
http://wiki.openwrt.org/doc/howto/build
-- or start here if unfamiliar with OpenWrt in general: http://wiki.openwrt.org

/ted
From: Ondrej Wisniewski 
Sent: Saturday, August 22, 2015 6:08 AM
To: ondrej1024/crelay 
Cc: Ted Hess 
Subject: Re: [crelay] crelay for OpenWRT (#3)

That was quick, thanks. Will inform you when a new version comes out.
Just one last question before I close this discussion. Where are the binary OpenWRT packages for crelay available?

ondrej1024, 2015-08-25:
@thess, I checked the snapshot repository for various architectures here: `https://downloads.openwrt.org/snapshots/trunk/[arch]/generic/packages/packages/` 
But I couldn't find the binary package for *crelay*. Shouldn't it be built by now?

thess, 2015-08-25:
I just checked the build logs and noticed a linking problem with hidapi when full-language support is enabled. I will fix this later today.

From: Ondrej Wisniewski 
Sent: Tuesday, August 25, 2015 2:52 AM
To: ondrej1024/crelay 
Cc: Ted Hess 
Subject: Re: [crelay] crelay for OpenWRT (#3)

@thess, I checked the snapshot repository for various architectures here: https://downloads.openwrt.org/snapshots/trunk/[arch]/generic/packages/packages/ 
But I couldn't find the binary package for crelay. Shouldn't it be built by now?

ondrej1024, 2015-09-02:
@thess: I checked and the crelay ipk files are built now. Great. Thanks for your support. Will close this issue now.

ondrej1024, 2015-09-02:
@jabbs: now you can use crelay by installing the official crelay packages from the OpenWRT development repositories. Let me know if that works for you.

closed 2015-09-02 by ondrej1024

jabss, 2015-09-15:
Hello,

Thanks for that.
Maybe I'm doing something wrong:

BusyBox v1.23.2 (2015-04-22 14:06:06 CEST) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 CHAOS CALMER (*15.05-rc1, r45695*)
 -----------------------------------------------------
  * 1 1/2 oz Gin            Shake with a glassful
  * 1/4 oz Triple Sec       of broken ice and pour
  * 3/4 oz Lime Juice       unstrained into a goblet.
  * 1 1/2 oz Orange Juice
  * 1 tsp. Grenadine Syrup
 -----------------------------------------------------

jabss@Caipirinha:~$ lsusb
Bus 001 Device 006: ID 0951:16a5 Kingston Technology
*Bus 001 Device 005: ID 16c0:05df Van Ooijen Technische Informatica HID
device except mice, keyboards, and joysticks*
Bus 001 Device 004: ID 067b:2303 Prolific Technology, Inc. PL2303 Serial
Port
Bus 001 Device 003: ID 0665:5161 Cypress Semiconductor USB to Serial
Bus 001 Device 002: ID 05e3:0608 Genesys Logic, Inc. USB-2.0 4-Port HUB
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
jabss@Caipirinha:~$ sudo opkg list-installed | grep relay
*crelay - 0.9-1*
jabss@Caipirinha:~$ /usr/bin/crelay
-ash: /usr/bin/crelay: not found
*jabss@Caipirinha:~$ /usr/bin/crelay -I*
*-ash: /usr/bin/crelay: not found*

*jabss@Caipirinha:~$ ls -larth /usr/bin/crelay-rwxr-xr-x    1 root     root
      26.9K Sep  8 09:59 /usr/bin/crelayjabss@Caipirinha:~$*

The crelay package was installed by downloading "crelay_0.9-1_ar71xx.ipk"
from trunk and with the command "opkg install ./crelay_0.9-1_ar71xx.ipk".
It donwloaded automatically a number of dependencies (which I cannot say
exactly which were)

Any suggestion?

Thanks,
João

thess, 2015-09-15:
I doubt that crelay will work on 15.05. CC packages are built with uClibc and on the trunk (DD) all packages are built with musl.