ondrej1024/crelay #4: Driver patch for kernel 3.18.12+

Schaussi, 2015-05-09:
Hi,

would you mind publishing a new driver binary for the current Raspbian kernel?
Thank you!

ondrej1024, 2015-05-11:
I might do this but I need some time. Last year I got my PC stolen with the Virtual Machine where I had the RPi kernel cross compile e environment installed. So I have to set this up again. Therefore it won't happen tomorrow or next week.
But maybe I could asked the RPi kernel maintainers, if they could include the cp210x patch in the kernel they release for the RPi. In this way I wouldn't have to build a kernel module for each new kernel version.
Ondrej

ondrej1024, 2015-08-14:
Hi Schaussi,
I have completely rewritten the Conrad card driver and now we don't need cp210x kernel driver any more. *crelay* talks directly to the card and it works great without driver patch.
It would be nice if you could try the latest version and send me your feedback. Thanks.

ondrej1024, 2015-08-21:
From version 0.8 onwards, the cp210x driver is not needed any more. Closing this issue.

closed 2015-08-21 by ondrej1024