ondrej1024/crelay #6: Sainsmart 16-channel with USB-HID

khilman, 2016-03-28:
This currently suports the Sainsmart 4- and 8-channel relays which use FTDI, but the Sainsmart 16-channel uses USB-HID, and looks different from the current hidapi driver.

Has anyone tried the Sainsmart 16-channel.  A very poor attempt at documenting them is on the Sainsmart page for this part:
http://www.sainsmart.com/mouse-over-image-to-zoom-sainsmart-16-channel-controller-usb-hid-programmable-control-relay-module-kit-sainsmart-16-channel-controller-usb-hid-programmable-control-relay-module-kit-sainsmart-16-channel-controller-usb-hid-programmable-contro.html

ondrej1024, 2016-03-29:
Looks interesting. Basically this is a small add on module with mini USB connector which drives the 16 output pins to control the relays on the card. Could also be used for other cards I guess.
Documentation is indeed very poor, but more than is usually supplied. I wonder what the HID_CMD_SIGNATURE is. 
I downloaded the zip file from the download link but it contains only Windows exe and dll files, pretty useless for us. But it also says 

> We will offer you a detailed project introduction and their source code(Contact us via support@sainsmart.com).

Could be worth a try.

khilman, 2016-03-29:
A little more experimentation and digging and I found out that the description about how to program these on the Sainsmart page is worthless.  Thankfully, someone used a protocol analyzer to reverse engineer the Windows driver and wrote a node.js utility: https://github.com/mvines/relay

I used that to hack up a quick C program (attatched below) using hidapi that is working fine for me.  With that, it should be easy to convert to a crelay driver.

[sain-usb-hid.c.txt](https://github.com/ondrej1024/crelay/files/194375/sain-usb-hid.c.txt)

khilman, 2016-04-08:
Implemented in pull request.  Closing.

closed 2016-04-08 by khilman