ondrej1024/crelay #10: Sainsmart 16 inputs card is recognised as 4/8

Kepedize, 2016-07-05:
I am running crelay on an Ubuntu machine.  I am replacing my conrad 4ch card with a sainsmart 16. when I run "crelay -i" i get: "Detected relay card type is Sainsmart USB 4/8-channel relay card (on FTDI chipid EECA5EC5, 8 channels)". I have tried both compiling from source and downloading the latest built binaries.

ondrej1024, 2016-07-05:
Unfortunately I don't have this card myself, so can't test.

@khilman: Do you have any idea? Is this card recognised correctly for you?

ondrej1024, 2016-07-05:
This is an idea. Maybe you have some other device with an FTDI chip connected. This would probably result in the Sainsmart USB 4/8-channel relay card being detected. Can you run the `lsusb` command on your machine and post the output here?

Kepedize, 2016-07-05:
Ok, that makes total sense, I do have a FTDI card (DEDITEC I/O).
lsusb output (quite a few):

_Bus 004 Device 002: ID 8087:8001 Intel Corp. 
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 8087:8009 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 0781:5591 SanDisk Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 009: ID 8087:07dc Intel Corp. 
Bus 001 Device 007: ID 0dd4:01a8 Custom Engineering SPA 
Bus 001 Device 005: ID 0416:5020 Winbond Electronics Corp. 
Bus 001 Device 003: ID 05a3:9520 ARC International 
**Bus 001 Device 014: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC**
Bus 001 Device 013: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 001 Device 012: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 011: ID 072f:223b Advanced Card Systems, Ltd 
Bus 001 Device 010: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 008: ID 0acd:2420 ID Tech 
Bus 001 Device 006: ID 1eab:0d03  
Bus 001 Device 004: ID 12d1:14ac Huawei Technologies Co., Ltd. 
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub_

ondrej1024, 2016-07-05:
OK, so this is the problem. Currently, crelay stops looking for relay cards when it finds the first one. Since your DEDITEC card has the exact same FTDI chip as the Sainsmart USB 4/8-channel relay card, crelay has no way of knowing which is which. 

You could change the order of the relay cards during detection so your 16 channel card is detected first. To do this just move the following lines of code to the beginning of the array `relay_data` in file relay_drv.c

```
#ifdef DRV_SAINSMART16
   {  // SAINSMART16_USB_RELAY_TYPE
      detect_relay_card_sainsmart_16chan,
      get_relay_sainsmart_16chan,
      set_relay_sainsmart_16chan,
      SAINSMART16_USB_NAME
   },
#endif
```

ondrej1024, 2016-07-05:
or build crelay without support for the Sainsmart USB 4/8-channel relay card:
`make DRV_SAINSMART=n`

Kepedize, 2016-07-05:
OK great, Thanks for the prompt help!

closed 2016-07-05 by Kepedize