ondrej1024/crelay #11: Suggestion for crelay.conf: State declaration (high = on, low = off, or vice versa)

Schaussi, 2016-07-23:
Example: Sainsmart 8-channel GPIO relay card. Relays are activated by output of LOW value (0) instead of 1. The result is, crelay web interface first activates every relay and then shows wrong states.

ondrej1024, 2016-08-07:
This would make perfect sense for all GPIO relay cards. We need an "active_low" option in the [GPIO drv] section of crelay.conf.
I will consider this feature for the next crelay version.

R1CH-T, 2017-07-27:
I'm currently working on adding an installation option for this great tool to [DietPi](http://dietpi.com). I have tested and found crelay to be working for my USB (2x relay) board, but in the in the process of testing it for my GPIO (2x relay) board, I have experienced this same weird behaviour.

When I launch Crelay as a daemon, nothing happens, but as soon as I open the web page for Crelay, the relays both turn on and when I click the **On** button, it turns them off.

I guess that a most people using these relay boards just control the GPIO pins directly and don't use Crelay, but I feel that this tool is a handy way of automating this and giving a cool little web interface, so thanks for your efforts.

However, I think that the behaviour described in this issue is a real problem and needs to be dealt with as soon as possible. The relays being controlled by Crelay are literally doing the exact opposite of what they are being told to do. In fact, by simply opening the web interface, all of the connected relays are being triggered. This is not good, to say the least!

>>This would make perfect sense for all GPIO relay cards. We need an "active_low" option in the [GPIO drv] section of crelay.conf.

>I will consider this feature for the next crelay version.

Although the addition of an "active_low" option would be great, this comment kind of misses the point: This is a serious bug, which needs to be addressed.

I'm sorry and I don't mean to offend; but, as I said this should be addressed as soon as possible.

I discovered Crelay through trawling the internet for a simple solution for controlling these USB relay cards. As such, it's totally brilliant and I commend you; but I have found the GPIO relay-control part to be problematic.

Please let me know your thoughts on this an if there's anything I can do to help. :-)
Please let me know

ondrej1024, 2017-07-30:
@R1CH-T I agree with you, this issue needs to be dealt with. I am currently on vacation but I hope to have a look at it next week.

ondrej1024, 2017-08-09:
I just started looking into this. Looks like there is a bit of a problem caused by the handling of the GPIO pins in the Linux Kernel. The GPIO pins are initialized to input mode by the kernel, so we have to set explicitly the pins we use for relay control to output mode. But the initial state of an output pin is 0 which will trigger the relays with inverted logic. We can set the pin to 1 immediately after defining it as output but I guess there will still be a small glitch. At the moment I really don't know how to avoid that.
Any ideas?

ondrej1024, 2017-08-15:
I think I found a good solution. When the crelay daemon starts, it initialises the GPIO pins as output and switches the relays off immediately. This is fast enough to not cause any glitches on the relay contacts.

@R1CH-T: can you please test the fix? It's on the [gpio_fix branch](https://github.com/ondrej1024/crelay/tree/gpio_fix). Just add `active_value = 0` in your config file for your active low logic relay card.

Thanks

ondrej1024, 2017-08-17:
@Schaussi : you opened this issue a long time ago but finally there is a solution. :smiley: 
Are you still interested in this feature and could you also test the implementation in the  [gpio_fix branch]( https://github.com/ondrej1024/crelay/tree/gpio_fix) ?

R1CH-T, 2017-08-17:
Hi @ondrej1024.

I'm really sorry for the delay, but I was away all last week and the internet connectivity was almost non-existent, so I didn't notice your reply from last Wednesday. I got back on Sunday, but just got around to dealing with my backlog of messages, this being at the top priority.

I will test this immediately and reply ASAP.

> I think I found a good solution. When the crelay daemon starts, it initialises the GPIO pins as output and switches the relays off immediately.

This is what I had in mind, but I would not have had the first idea of how to implement it. :)

> This is fast enough to not cause any glitches on the relay contacts.

Fantastic. It passed through my mind that this might be a problem.

> @R1CH-T: can you please test the fix? It's on the gpio_fix branch. Just add active_value = 0 in your config file for your active low logic relay card.

I'm on it! :)

Schaussi, 2017-08-21:
Hi ondrej,
this sounds very good :)
Because I opened this issue a quite long time ago, I am not able to test it at the moment, sorry.
Thank you!

R1CH-T, 2017-08-21:
Sorry @ondrej1024, but I was not able to test the new implementation immediately, like I said, but I have been extensively testing this morning and can say that it works brilliantly!

I found that the Raspberry Pi must be rebooted for changes to the config file to take effect without any issues, but apart from that, all is working fine.

I have created a couple of video files, showing a demonstration of Crelay on a 8 relay GPIO board; one for **active_value = 0** and one for **active_value = 1**.

Here they are:

 **active_value = 0:**
[image]

**active_value = 1**
[image]

Thanks for the great work!
Rich.

P.S. Is there any way to control more than 8 relays via Crelay? I ask because I've also got one of these monsters:

[[image]

R1CH-T, 2017-08-21:
Sorry for taking so long to test this, when I said I'd do it right away, but I have performed extensive testing this morning and found that it works brilliantly.

Thank you so much @ondrej1024!

Here are links to two videos showing Crelay in action on my 8-relay board:

https://vimeo.com/230428358

https://vimeo.com/230431557

ondrej1024, 2017-08-21:
Thanks for testing @R1CH-T . I'm glad things are working fine. Nice to see this in action in the video.
So I will merge this to the master branch and put out a new release ASAP.

I didn't think a reboot of the RPi is needed after changing the configuration but just a restart of crelay. I will have another look at this.

To support 16 relays the definition of `GENERIC_GPIO_NUM_RELAYS` in `relay_drv.h` has to changed to 16 and all data structures holding the GPIO pin data need to extended. I don't think there is much more to do to make this work.

R1CH-T, 2017-09-06:
Hey @ondrej1024.

Once again, sorry for the long delay in getting back to you about this.

I will be happy to help in any way I can in testing any new implementation of Crelay and provide evidence of it working as I did before; just let me know.

If you want to let me know which are the files, etc, that need changing, I can take a look, but I might be a little out of my depth...

Thanks again for your work on this project so far.
 Rich.
:100:

ondrej1024, 2017-09-18:
Sorry for the delay, had other priorities these days, but I hope to get the new official release out in the next few days.

ondrej1024, 2017-11-03:
@R1CH-T, I just released this fix to the master branch. 
As for the patch allowing 16 GPIO relays, I need some more time for testing.

R1CH-T, 2017-11-08:
Thanks so much @ondrej1024!

I look forward to the next patch.

closed 2017-11-10 by ondrej1024