ondrej1024/crelay #12: Unable to Detect Relay Status (PiRelay app)

szygmunt, 2016-08-10:
I have 2 channel HID interface relay board. It's successfully running under openwrt in daemon mode but I have problem with controlling it under PiRelay android app. Application have a problem with checking relay status after start and always sets relays as OFF. Is this a problem in PiRelay or my crelay configuration?

ondrej1024, 2016-08-11:
Are you using the free or the pro version of PiRelay? The free version does not check the actual relay status but initialises and shows all buttons as OFF.
To test the HTTP API of crelay you can open the following URL in your browser:
`http://ip_address[:port]/gpio`

As response you should get this (with the real relay states: 0 or 1)
```
Relay 1:0
Relay 2:0
```
The Pro version of PiRelay should request and read this information on startup. But I have never tried the Pro version myself.

ondrej1024, 2016-08-11:
I implemented the format of the HTTP API to be compatible with PiRelay. That means crelay sends the same format as does the native Python code of PiRelay that runs on the server. This is documented [here](https://sites.google.com/a/jasonfindlay.com/pi-relay/). The related Python code reported there is the following, which should produce exactly the same result as the output from crelay:
`echo "Relay 1:".$output."<br>";`

So I don't know why PiRelay Pro doesn't detect the correct relay states. As I said, I haven't tested this because I never bought the Pro version.

szygmunt, 2016-08-11:
Thank you. So i have to ask author of PiRelay about that.

ondrej1024, 2016-08-11:
Anyway I am not happy with this situation. It look like the free version of PiRelay was removed from the Play store and I don't want a user of crelay to be forced to pay money for an app to be used with crelay. For this reason and the fact that PiRelay is not open source I am considering to provide my own app for crelay. Unfortunately I have to find some time for this which looks difficult at the moment. But maybe there are some volunteers :smirk:

ondrej1024, 2016-08-11:
I had another look at the PiRelay Python code and I actually see some differences between the output it generates and the one generated by crelay. 
The original code always generates output lines for 8 relays. crelay only generates the lines for the relays that actually exist. And the original code also generates one first character which reports the value of the 'status' parameter from the request.
`echo $_GET['status'];`
So probably the PiRelay app expects exactly this kind of data and gets confused if the character isn't there or not all 8 relay states are reported.

szygmunt, 2016-08-11:
Dnia 11 sierpnia 2016 14:50:36 CEST, Ondrej Wisniewski <notifications@github.com> napisał(a):
> I had another look at the PiRelay Python code and I actually see some
> differences between the output it generates and the one generated by
> crelay.
> The original code always generates output lines [...]

Could you make some patch for test?
-- 
Szymon Zygmunt

ondrej1024, 2016-08-11:
Please try the attached patch and let me know if it works.
[crelay-pirelay-patch.txt](https://github.com/ondrej1024/crelay/files/413783/crelay-pirelay-patch.txt)

szygmunt, 2016-08-11:
Patch is ok. PiRelay get relays status properly.

ondrej1024, 2016-08-12:
Thanks for testing. I'm glad it's working. 
What do you mean  by _without "for"_? Are you referring to the second "for" loop I added to generate the remaining relay state output for the relays that are not present?

szygmunt, 2016-08-12:
Yes I'm talking about this loop:
`+      for (i=last_relay+1; i<=8; i++)
+      {
+         fprintf(f, "Relay %d:%d<br>", i, 0);
       }`
which isn't required to compatibility crelay and PiRelay.

ondrej1024, 2016-08-12:
OK, good to know. Then it was probably just the HTTP header which was missing to send a correct HTTP response to the app. I will include this patch in the next crelay release, which will be a bugfix release.

szygmunt, 2016-08-12:
Maybe will you improve web server code which we were talking on openwrt github [https://github.com/openwrt/packages/issues/3016]

szygmunt, 2016-08-12:
Last test and I can confirm that only HTTP header was a reason for lack of cooperation crelay witch PiRelay.

ondrej1024, 2016-08-12:
I looked also at the web server code and it is not trivial to remove the file stream handling as suggested by @thess. I think I will just integrate his patch in the next release and do the major rework at a later time.

szygmunt, 2016-08-12:
Ok. Thanks.

ondrej1024, 2016-08-12:
@thess, I updated crelay.c with some fixes (including the issue discussed here and your file stream patch). Could you please test this and let me know if it is now running on OpenWRT without further patching?
Thanks.
[crelay.c.txt](https://github.com/ondrej1024/crelay/files/415258/crelay.c.txt)

ondrej1024, 2016-08-30:
Released [V0.10.1](https://github.com/ondrej1024/crelay/releases/tag/V0.10.1) which includes the patch discussed above.

closed 2016-08-30 by ondrej1024