ondrej1024/crelay #16: 3d printer as a relay

rommaster, 2017-11-01:
Hi, i’m new to all this linux stuff, but the problem is that crelay sees my 3d printer as sainsmart relay and my printing software doesn’t see printer when crelay is on. How can a fix this problem? I use orange pi on armbian

ondrej1024, 2017-11-02:
I suppose you have both the relay card and 3d printer connected to your system.
What brand is your relay card you want to control? Can you send me the output of this command ?
`crelay -i`

rommaster, 2017-11-02:
I don’t have the relay yet. I removed crelay for now, but as i remember crelay -i gave me something like “ sainsmart 4-8 channel usb relay 00000000”. As i have understood instead of 00000000 there should be serial number.

rommaster, 2017-11-02:
https://www.aliexpress.com/item/2-Channel-USB-Relay-Module-Programmable-Computer-Control-For-Smart-Home-DC-5V/32782208617.html?ws_ab_test=searchweb0_0,searchweb201602_2_10152_10065_10151_10068_10344_10345_10342_10343_10340_10341_10541_10540_10307_10303_10060_10302_10155_10154_10056_10055_10054_10539_10538_10537_10059_10536_10534_10533_100031_10099_10103_10102_5590020_10052_5640020_10053_10107_10050_10142_10051_10170_10084_10083_5370020_10080_10082_10081_10110_10111_10112_10113_10114_10312_10313_10314_10078_10079_10073,searchweb201603_30,ppcSwitch_2&btsid=f340d24a-14a4-4532-9226-49e02183e2c3&algo_expid=7fee07c8-eccc-4939-8653-955ebbad4505-0&algo_pvid=7fee07c8-eccc-4939-8653-955ebbad4505       I guess this is noname chinese relay

ondrej1024, 2017-11-02:
Apparently your 3d printer uses the FT245RL chip for the USB interface, which is the same as the Sainsmart relay card. This means crelay cannot tell the difference. 

You can build crelay without the Sainsmart driver:
`make DRV_SAINSMART=n`

Then crelay will not see your 3d printer anymore, but it will also not work with the Sainsmart relay card. So you should buy a different relay card type which is supported by crelay.

ondrej1024, 2017-11-02:
Yes, this should be the HID API compatible relay card, so it will work with crelay and the Sainsmart driver disabled. Then there will be no conflict with the 3d printer.

rommaster, 2017-11-02:
Thank you for help

rommaster, 2017-11-10:
Thanks to you everything works like it should be, i’m closing the issue.

closed 2017-11-10 by rommaster