ondrej1024/crelay #17: I have issue with turning my relay on and off.

Bhanushree2904, 2017-11-20:
Hi, 

I am new in using relays, I see this  https://www.raspberrypi.org/forums/viewtopic.php?t=110710 Please Help Noobs control 8 channel relay post and issues regarding it. 
I have everything installed and I have connected my raspberry pi to a 8 channel usb relay. The device is detected. 

pi@raspberrypi ~ $ sudo crelay -i
Detected relay cards:
#1 Sainsmart USB 4/8 channel relay card (serial AI04XQI3)
pi@raspberrypi ~ $sudo crelay 1 on
pi@raspberrypi ~ $ 

I have it up to the above point. I cannot turn on the relay or turn it off. I get no output!!! Please help. Thanks in advance.

Bhanushree2904, 2017-11-20:
I have this working after reconnecting the power source to my relay!! Thanks :)

closed 2017-11-20 by Bhanushree2904

ondrej1024, 2017-11-20:
> I have this working after reconnecting the power source to my relay!!

That's what I thought :smiley:

Anushree2904, 2017-11-21:
Hi, 

I have a quick question. Can I control the relays to on and off for certain number of seconds from the command line? using  $ sudo crelay 1 on method ? 

Thanks, 
Bhanu

ondrej1024, 2017-11-22:
No, this feature is not implemented from command line, only for Web GUI. To use this function from command line we could implement something like this:
` $ sudo crelay 1 pulse`

If you know to code in C you could provide a patch with this feature. Otherwise just open a new issue with this feature request.