ondrej1024/crelay #18: Installing to Debian Stretch

dlewin, 2017-12-15:
libudev won't install until you : 
- sudo apt-get install pkg-config libusb-dev

ondrej1024, 2017-12-15:
libudev should only be needed if you build the HIDAPI lib from source. But in recent Debian releases the [libhidapi-dev package](https://packages.debian.org/stretch/libhidapi-dev) can be installed from the Debian repos.
Did you try this?

dlewin, 2017-12-15:
Actually it was for the doc. part. Thus following the provided steps I have installed :

- libftdi1
- libhidapi-libusb0
- libusb-1.0-0
- libftdi1
- libftdi-dev
- libhidapi-libusb0
- libhidapi-dev
-  libusb-1.0-0
- libusb-1.0-0-dev
- libudev-dev

and the configure was not possible until : 

- dh-autoreconf
- pkg-config libusb-dev

ondrej1024, 2017-12-15:
You don't need to run `configure` at all. This is part of the optional installation step "Build dependencies from source" which should not be needed on Debian Stretch if the previous step "Install dependencies" was successful. 
I think I will just delete the optional step from the doc as most people will not run Debian Wheezy or older anymore.

dlewin, 2017-12-15:
ok

closed 2017-12-15 by dlewin