ondrej1024/crelay #28: segfault after disconnect

peterdedecker83, 2018-10-26:
Hi all,
I have a sainsmart 4 relay board, with these applications:
- relay 1: not connected
- relay 2: garage door pulse button
- relay 3: 240V light socket with two led bulbs
- relay 4: 240V magnetic water valve

Everything works fine, except relay 4. When I activate the relay through the web interface, the relay gets activated correctly and the page is reloaded as it should be. However, when I click to deactivate the relay, the daemon terminates with this in /var/log/messages:

```
Oct 26 20:23:39 localhost vmunix: [ 6640.879106] usb 2-1: USB disconnect, device number 4
Oct 26 20:23:39 localhost vmunix: [ 6641.143125] usb 2-1: new full-speed USB device number 5 using ohci-pci
Oct 26 20:23:39 localhost vmunix: [ 6641.335615] usb 2-1: New USB device found, idVendor=0403, idProduct=6001
Oct 26 20:23:39 localhost vmunix: [ 6641.335994] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 26 20:23:39 localhost vmunix: [ 6641.336475] usb 2-1: Product: FT245R USB FIFO
Oct 26 20:23:39 localhost vmunix: [ 6641.343317] usb 2-1: Manufacturer: FTDI
Oct 26 20:23:39 localhost vmunix: [ 6641.350241] usb 2-1: SerialNumber: AI02JFXJ
Oct 26 20:23:39 localhost vmunix: [ 6641.371600] crelay[2861]: segfault at 40001fd ip b767cee6 sp bf8ca8c0 error 4 in libc-2.19.so[b7607000+16e000]

```

So it seems the relay board gets disconnected and connected again. Maybe there is kind of an issue with the board itself, which is weird because everything worked fine before when relay 4 was used for a light socket.

When I reload the daemon, all works fine again until I deactivate relay 4 again, after which the relay is correctly deactivated but the daemon terminates again due to this disconnection issue.

I use crelay 0.13. The CLI doesn't work either:

```
root@alix:~# crelay -i 3

Detected relay cards:
  #1	Sainsmart USB 4/8-channel relay card (serial AI02JFXJ)
```

Thanks!

peterdedecker83, 2018-10-26:
Even if it's not possible to solve the disconnection issue (it might be a fault in the board), it still would be nice if the daemon just ignores the disconnection of the board and waits for a re-connection of the board.

ondrej1024, 2019-01-03:
Hi @peterdedecker83 , did you solve your problem ?

peterdedecker83, 2019-01-03:
Hi Ondrej, I installed Monit and your Monit script to restart the service when crashed. This is a suboptimal solution, but works smoothly for me now. Thanks.

ondrej1024, 2019-01-07:
Thanks for the update. So we have a workaround at the moment. 
However, I should probably implement better error checking in the relay driver to avoid the segfault. But I need to find out exactly what instruction in the code causes the fault. I need some time to do this.

peterdedecker83, 2019-01-07:
Indeed, a proper functioning workaround.

ondrej1024, 2021-12-13:
Finally :smile:  this bug was fixed with commit 29b37c0065c0eb892e1d8b1a6efacd63cee5501e
Closing the issue.

closed 2021-12-13 by ondrej1024