ondrej1024/crelay #29: Suggestions for Multiple Relay Hardware?

NadavK, 2018-11-13:
Can anyone suggest hardware for a total of 48 relay outputs?
Another thing, it needs to handle 16A @ 250VAC (at least for some of the relays)

I apologize if this is not the correct forum/channel for questions like this.

ondrej1024, 2018-11-15:
From the relay cards I have tested you could use 6 [Sainsmart 8 channel cards](https://www.sainsmart.com/products/8-channel-12v-usb-relay-module) or 3 [16 channel cards](https://www.sainsmart.com/products/16-channel-usb-hid-programmable-control-relay-module).
However the relays on these cards have a maximum rating of 10A @ 250VAC.

NadavK, 2018-11-15:
Thanks for the suggestion!
Does issue https://github.com/ondrej1024/crelay/issues/26 prevent using multiple cards?

ondrej1024, 2018-11-15:
Yes, but wit the patched HIDAPI from @derekatkins it works, as explained in the related discussion.

NadavK, 2018-11-15:
The 16 channel model you linked to is no longer available.
Would [this 16 channel model](https://www.sainsmart.com/products/16-channel-9-36v-usb-relay-module) be compatible?
(hope I am not abusing your helpfulness)

ondrej1024, 2018-11-15:
> The 16 channel model you linked to is no longer available.
> Would [this 16 channel model](https://www.sainsmart.com/products/16-channel-9-36v-usb-relay-module) be compatible?

This relay card is not yet supported by `crelay`. We have discussed this with @kjatar in issue #21 . It should be possible to implement support for this card but I need someone to test is as I don't have the card myself (and no time at the moment :cry:  )