ondrej1024/crelay #30: Multiple relay control?

ldnelso2, 2018-12-23:
Hi - maybe a stupid request, but I can't seem to get multiple relays to fire at the same time (not multiple boards, just multiple relays on a multi-channel relay)

This seems to work fine for me: 
curl -d "pin=3&status=1" http://localhost:8000/gpio

but posting multiple variables doesn't seem to work no matter what syntax I have tried so far.

ondrej1024, 2018-12-23:
In fact this is not supported. You have to switch one relay at a time.

In the future I would like to change the API and use json format for the requests. Then it would probably be quite straight forward to implement handling of multiple relays in one request.

ondrej1024, 2018-12-23:
Not really. Crelay supports many kind of relay cards and every relay card works differently. There is a low lever driver for each card type which implements the specific communication protocol. 

However the user interface or API hides the specific cards implementation, so you don't have to worry about this and every card is controlled the same way. The problem is that the current API syntax makes it difficult to specify multiple relays in the same request. 

This could be nicely done in json format though, e.g. the request parameters to switch 3 relays at the same time could look like this:
`request=[{"relay":1,"state":"off"},{"relay":2,"state":"on"},{"relay":4,"state":"off"}]`

But for this to work the API handling of `crelay` has to be completely changed. Check `crelay.c` to see how it is currently implemented.

ondrej1024, 2018-12-23:
Internally, after having parsed the API request, `crelay` would still switch one relay at a time, but it would be really quick and seem almost at the same time?

What is actually your goal? Do you want to minimise the API requests or do you actually really need to switch multiple relays at the same time ?

ldnelso2, 2018-12-23:
I see, that makes sense.

I need to switch multiple relays on and off at the same time across
multiple boards simultaneously.  I have 256 relays that need to fire at
once (and not all on or all off, uniques combinations).

ldnelso2, 2018-12-23:
I will probably use sixteen of the 16 channel sainsmart relays unless I can find a better option.

ldnelso2, 2018-12-24:
Hi Ondre - it is a manufacturing automation application.  Much faster to have all the multiple actuators ready to go simultaneously than go 'round robin' to just the spots which require 'on.'

I looked a bit more into the drivers, and checked out darrylb123's work.  Was able to get multiple relays firing at the same time without too much trouble.  I will keep you posted on what we do.  I don't mind helping contribute, but I don't know c very well.  Best, and happy holidays!

ldnelso2, 2019-01-14:
Hi Ondre - 

It turns out the Sainsmart relay I had isn't HID compatible.  Some of the sainsmart 16-channel USB relays use an HID compatible interface, and some don't.  Mine unfortunately didn't have a linux driver either, but they did have serial commands, so I adapted a python script from online to control it.  I am not much of a C programmer, otherwise would have done it that way.

I bought two of them and eventually got them working but struggled to get status messages back and there was some whining noise coming out of one of the relays.   Given that my application needs quite a few relays (but maybe not enough to go with a costly PLC approach), we are looking at relay cards with ethernet ports.  

Thank you for answering my initial question and I am going to go ahead and close this out.  

Even though we probably won't use these cards, it was challenging to find the wiki page with the serial commands at first (it isn't mentioned in the packaging or on the product page), and they don't have a linux driver (only windows).  I will create a new issue and post my python code for controlling the 16-channel QinHeng Electronics HL-340 USB-Serial adapter as a separate issue in case you want to support that card in the future and it may be of help.

closed 2019-01-14 by ldnelso2