ondrej1024/crelay #31: Sainsmart 16-channel - QinHeng Electronics HL-340 USB-Serial

ldnelso2, 2019-01-14:
I purchased a Sainsmart 16-channel USB relay.   There is support for the Sainsmart 16-channel relay within crelay, but upon further inspection after receiving the card, the card that I purchased (SKU: 101-70-208) is not HID compatible, and instead shows up as:  QinHeng Electronics HL-340 USB-Serial.  I could be wrong, but I don't think this variety is supported yet by crelay.

After finally finding the serial commands under the sainsmart wiki for the part number, I adapted a python script from pyusb to get it working initially, although I couldn't get the board to return a status (and I purchased two boards), at least the relays would turn on and off: 

https://github.com/ldnelso2/sainsmart

ondrej1024, 2019-01-14:
There is an ongoing discussion in Issue #21 about support for these Sainsmart cards. We have collected some documentation but there are basically 2 things missing:
* Information on how to read the status from a card
* Information on how to read specific serial number or Id from each card (for multiple card support)

ldnelso2, 2019-01-14:
Okay, sorry I looked but didnt see that one.  I wasn’t able to get either
to work.  I do not think the device is serialized, and I was not able to
get the status returned from the board, thus omitted status check from my
code, even though sainsmart provided the serial commands... they don’t
work.

closed 2019-05-09 by ldnelso2