ondrej1024/crelay #32: Get this to work on Standard linux

lseg74, 2019-03-13:
Hi,

Will this project only work in a OpenWrt environment. Or can I get this to work on a standard Linux PC.
If yes, how?
I am somewhat missing How to get this to work for me. And I really would like to get this to work.

Kind regards,

Luc

ccutrer, 2019-03-13:
I have it running on Ubuntu bionic, amd64 machine. I just followed the Installation From Source instructions in the readme - basically install all the dependencies, download the code, `make`. it has to be run as root, so I also set the setuid bit so that I could call it from less-privileged code.

ondrej1024, 2019-03-15:
> I am somewhat missing How to get this to work for me.

Hi Luc, did you read the documentation? As @ccutrer wrote, everything is described in the README. 

For OpenWrt there are ready to use packages so it is very simple to install crelay from their repositories, but for all other platforms you need to build the software as described.

ondrej1024, 2019-03-16:
> Is this also enough to get the HTTP api working?

Yes, HTTP api is always enabled when running `crelay` in daemon mode.

closed 2019-03-30 by ondrej1024