ondrej1024/crelay #33: Permission issue?

MajorLOLGaming, 2019-04-05:
Hello. I am still in the early process of learning all things for raspberry pi, and has just discovered this amazing thing. I got it set up and web interface is working. I just have one problem. when i switch the relay(s) on, in my web interface (Probably the same if i use the POST or GET requests.), it gives me an error.
`ERROR: Unable to pwrite 1 to gpio value: Operation not permitted`
and the same with when i switch it off, of course.
`ERROR: Unable to pwrite 0 to gpio value: Operation not permitted`

This is probably a easy fix. Any help is accepted. :)

ondrej1024, 2019-04-06:
Hi, I guess you don't use an USB relay card but relays via GPIO pins.

Are you running crelay as user `root` ?
Did you specify the GPIO pins where your relays are connected in `crelay.conf` ?

MajorLOLGaming, 2019-04-06:
Hello. I will try to run it as root soon. crelay.conf has been correctly set up. And yes, I use gpio pins. Thanks.

MajorLOLGaming, 2019-04-07:
nope. running as root did not work

MajorLOLGaming, 2019-04-07:
Update: I got it to turn on the relay. Now i can't get the web interface to send requests to the server... When i toggle it, on the web, i get no output in the console.

ondrej1024, 2019-04-07:
> Update: I got it to turn on the relay.

What did you do to make it work?

>  Now i can't get the web interface to send requests to the server... When i toggle it, on the web, i get no output in the console.

What do you expect to see on the console? You must be running `crelay` in daemon mode. Toggling the switch on the web interface will toggle the relay state on/off, that's it.

Please explain what exactly **is** working and what **is not**.

MajorLOLGaming, 2019-04-07:
To make it turn on the relay:
I mixed up the pin numbers. My bad.

What is working:
When i start crelay with `sudo crelay -d` it starts normally, enabling the web interface on port 8000.

What is not working:
Toggling the switch on, does nothing, thereby outputting nothing in the console, and does not toggle the relay.

What happens when i execute `sudo crelay -d`:
Relay toggles on, web interface launches, console gets ready to display activities. I toggle the switch, nothing happens. No relay toggle, no activity in console.

ondrej1024, 2019-04-07:
You are probably still using the wrong pin numbers. You have to specify the **BCM pin number**, not the physical or the wiringPi pin numbers. Check [here](https://pinout.xyz) for the correct numbers. For example, if your relay is connected to physical pin 11, you have to use the BCM pin number 17 in `crelay.conf`.  

Also check you have correctly configured the `active_value`:
`
active_value = 1       # 1: active high, 0 active low
`

MajorLOLGaming, 2019-04-07:
I've done that. The active value is 1 and the BCM pin I use is 26.

ondrej1024, 2019-04-08:
OK, that means you have connected your relay to physical pin 37, right?
You can check the content of the following directory:
`/sys/class/gpio/`

There should be the subdirectory `gpio26` created which contains the file `value`. This file contains the value of your IO pin (0/1). Please check if this value changes when you toggle the relay from the web GUI.

Please also post your complete `crelay.conf` here.

MajorLOLGaming, 2019-04-08:
That is correct. 
When i get home i will do that. Thanks!

MajorLOLGaming, 2019-04-08:
Hello, `/sys/class/gpio/gpio26/` for some reason don't seem to be a directory. Should i just create the directory and the file manually?
Fyi i am using the Raspbian Stretch Lite os.

Here is my `crelay.conf`:
`################################################
#
# crelay config file
#
# This file is read by crelay in daemon mode
# from /etc/crelay.conf
#
################################################

# HTTP server parameters
################################################
[HTTP server]
server_iface = 0.0.0.0    # listen interface IP address
#server_iface = 127.0.0.1 # to listen on localhost only
server_port  = 8000       # listen port
relay1_label = LED   # label for relay 1 (Note: (This is not in the crelay.conf) I plan on using this for my leds)
relay2_label = Device 2   # label for relay 2
relay3_label = Device 3   # label for relay 3
relay4_label = Device 4   # label for relay 4
relay5_label = Device 5   # label for relay 5
relay6_label = Device 6   # label for relay 6
relay7_label = Device 7   # label for relay 7
relay8_label = Device 8   # label for relay 8

# GPIO driver parameters
################################################
[GPIO drv]
num_relays = 1    # Number of GPIOs connected to relays (1 to 8)
active_value = 1       # 1: active high, 0 active low
relay1_gpio_pin = 26   # GPIO pin for relay 1 (26 for RPi GPIO37)
#relay2_gpio_pin = 18   # GPIO pin for relay 2 (18 for RPi GPIO1)
#relay3_gpio_pin = 27   # GPIO pin for relay 3 (27 for RPi GPIO2)
#relay4_gpio_pin = 22   # GPIO pin for relay 4 (22 for RPi GPIO3)
#relay5_gpio_pin = 23   # GPIO pin for relay 5 (23 for RPi GPIO4)
#relay6_gpio_pin = 24   # GPIO pin for relay 6 (24 for RPi GPIO5)
#relay7_gpio_pin = 25   # GPIO pin for relay 7 (25 for RPi GPIO6)
#relay8_gpio_pin = 4    # GPIO pin for relay 8 ( 4 for RPi GPIO7)

# Sainsmart driver parameters
################################################
[Sainsmart drv]
# num_relays = 4   # Number of relays on the Sainsmart card (4 or 8)`

ondrej1024, 2019-04-08:
> Hello, `/sys/class/gpio/gpio26/` for some reason don't seem to be a directory. Should i just create the directory and the file manually?

It is not a directory but a symbolic link to a directory. Anyway You should be able to cd into `gpio26` and then see the `value` file. If this is not the case there is something very weird going on. What is the output of 
`ls -l /sys/class/gpio/`

These gpio directories are created by the kernel when you write the pin id to `/sys/class/gpio/export` . This is actually what crelay does to control the GPIO pin. If you are interested you can read more about it [here](https://falsinsoft.blogspot.com/2012/11/access-gpio-from-linux-user-space.html).

MajorLOLGaming, 2019-04-08:
Oh okay. I get it.
This is the output.
```total 0
-rwxrwx--- 1 root gpio 4096 Apr  7 21:19 export
lrwxrwxrwx 1 root gpio    0 Apr  7 21:19 gpiochip0 -> ../../devices/platform/soc/20200000.gpio/gpio/gpiochip0

-rwxrwx--- 1 root gpio 4096 Apr  7 21:19 unexport```

ondrej1024, 2019-04-08:
Where is the `gpio26` folder? Is crelay not running?
Please start crelay as root and then check `/sys/class/gpio/` again. Also check the syslog just after you launched crelay with `tail -50 /var/log/syslog`.
Did you copy your `crelay.conf` into `/etc/` ?

MajorLOLGaming, 2019-04-08:
ls -l /sys/class/gpio/ (This time with crelay running): 
```
total 0
-rwxrwx--- 1 root gpio 4096 Apr  7 21:19 export
lrwxrwxrwx 1 root gpio    0 Apr  8 22:24 gpio26 -> ../../devices/platform/soc/20200000.gpio/gpiochip0/gpio/gpio26
lrwxrwxrwx 1 root gpio    0 Apr  7 21:19 gpiochip0 -> ../../devices/platform/soc/20200000.gpio/gpio/gpiochip0
-rwxrwx--- 1 root gpio 4096 Apr  7 21:19 unexport
```

tail -50 /var/log/syslog:
```
Apr  8 22:30:20 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:30:34 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:30:50 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:05 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:20 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:35 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:43 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:47 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:50 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:31:53 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:32:05 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:32:20 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:32:27 pizero systemd[1]: Started Session c5 of user pi.
Apr  8 22:32:35 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:32:42 pizero systemd[1]: Started Session c6 of user pi.
Apr  8 22:32:50 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:32:57 pizero hyperiond[501]: EFFECTENGINE INFO: effect finished
Apr  8 22:32:57 pizero hyperiond[501]: BORDER SWITCH REQUIRED!!
Apr  8 22:32:57 pizero hyperiond[501]: CURRENT BORDER TYPE: unknown=0 hor.size=0 vert.size=0
Apr  8 22:33:05 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:33:20 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:33:35 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:33:59 pizero systemd[1]: Started Session c7 of user pi.
Apr  8 22:34:05 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:34:15 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:34:20 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:34:24 pizero crelay[3027]: Starting crelay daemon (version 0.14)
Apr  8 22:34:24 pizero crelay[3027]: Config parameters read from /etc/crelay.conf:
Apr  8 22:34:24 pizero crelay[3027]: ***************************
Apr  8 22:34:24 pizero crelay[3027]: server_iface: 0.0.0.0
Apr  8 22:34:24 pizero crelay[3027]: server_port: 8000
Apr  8 22:34:24 pizero crelay[3027]: relay1_label: LED
Apr  8 22:34:24 pizero crelay[3027]: relay2_label: Device 2
Apr  8 22:34:24 pizero crelay[3027]: relay3_label: Device 3
Apr  8 22:34:24 pizero crelay[3027]: relay4_label: Device 4
[... cut ...]
Apr  8 22:35:05 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:35:20 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
Apr  8 22:35:35 pizero dhcpcd[271]: wlx801f0211ce4f: Router Advertisement from fe80::a263:91f                                                                                                f:feff:5bb3
```

Yes. The location of crelay.conf is  `/etc/crelay.conf`

ondrej1024, 2019-04-08:
Now do this:
`cat /sys/class/gpio/gpio26/value`
and check the output.

Then toggle the relay in the WebGUI and do it again. Does the value change?

MajorLOLGaming, 2019-04-08:
No. it doesn't. :/

ondrej1024, 2019-04-08:
Unfortunately I don't have any other ideas. Don't know what the problem is. Works fine here.

MajorLOLGaming, 2019-04-08:
hmm... i will try and reset my system tomorrow. thanks for trying :)

MajorLOLGaming, 2019-04-09:
Good news! It's working again after a complete reset of my system! Only one issue. The relay is not toggleing, when i toggle it in the web interface. `crelay.conf` is the same, and `cat /sys/class/gpio/gpio26/value` now changes value when i toggle it. I feel like its close to working!

ondrej1024, 2019-04-09:
You might have wired your relay incorrectly. Check the GPIO pin with a multimeter. When `value` is 1 you should measure 3.3V between the pin and GND, when `value` is 0 you should measure 0V.

MajorLOLGaming, 2019-04-09:
I've ordered it. Thanks for the help.

closed 2019-04-09 by MajorLOLGaming