ondrej1024/crelay #34: IFTTT?

MajorLOLGaming, 2019-04-14:
Is it possible to make this work with ifttt? As in, "If **You say "turn on relay"**, then **Make a web request**" with the webhook applet. Thanks.

ondrej1024, 2019-04-15:
I don't know. But I assume that you are talking about the [IFTTT](https://ifttt.com/) cloud service. So I am pretty sure you can find all the information from their documentation. 
Please post all your findings here as it could be helpful for others (including myself).

MajorLOLGaming, 2019-04-15:
Okay. I will try and test it when I get a new pi. Long story short, I accidentally fried one. It was only a pi zero, so it didn't cost that much to replace

MajorLOLGaming, 2019-04-17:
Got my new pi now. I will begin experimenting now.

MajorLOLGaming, 2019-04-17:
YES! I did it! Here's how to do it:

1. Setup your crelay, and make sure it's working correctly.
2. Forward the crelay port (By default it's 8000, though i recommend changing it, if you have shared your IP with anyone.) through your router. ([This guide can be helpful](https://www.noip.com/support/knowledgebase/general-port-forwarding-guide/))
3. Find your public IP, eventually by using [WhatIsMyIPAddress](https://whatismyipaddress.com/), and looking under IPv4. 
4. Go to [IFTTT](https://ifttt.com/), and create an account.
5. Create a new applet, by going under "[My Applets](https://ifttt.com/my_applets)", and then pressing the "[New Applet](https://ifttt.com/create)" button.
6. Press the "+this", and choose a service, which in this case, will be Google Assistant.
7. Then choose a trigger, (I chose the one called "Say a simple phrase".) and set it up to your likings.
8. When you have set up your trigger, press the "+that", and choose the service called "Webhooks", then the action called "Make a web request".
9. Under "URL" you have to put in "https://**_YourPublicIP_**:**_YourcrelayPort_**/gpio?pin=**_RelayNumber_**&status=**_Status_**", with **_YourPublicIP_** being the IP you got from [WhatIsMyIPAddress](https://whatismyipaddress.com/), and **_YourcrelayPort_** being the port you use for crelay, and **_RelayNumber_** being the relay that you wan't to turn on, and **_Status_** being "1" for "On" and "0" for "Off".
10. The method should be "GET", and the Content Type should be "text/plain", and the Body should be left empty.
11. Now press "Create action", and then "Finish" when redirected.
12. Now repeat the steps from 5 to 11, only with the "1" in "&status=1" being a "0"

Sorry if this was a little hard to understand, as im still a little hyped after getting it to work. Have fun! :)

closed 2019-04-17 by MajorLOLGaming