ondrej1024/crelay #36: Crelay crash

Sztoppolino, 2019-07-18:
Hi!

I have a problem. I use crelay on my OpenWRT router. It succesfully installed and it control my 1-CH relay card properly. 

Unfortunately, crelay always crash after undefinied time (~1 minute - ~1 day). Sometimes a fast router-reboot is enough, but usually I have to re-install the package.

This issue is very weird, because I can't find any reasons. It happens randomly. When I use (switch on/off) the relay, suddenly the web UI disappear, and my web broswer can't estabilish the connection with crelay anymore. Moreover, PiRelay neither. It sends me an error message:

"Pi (server) not detected on network"

Do you have any ideas (or solutions?)

Thank you in advance! ;)

ondrej1024, 2019-07-19:
This is probably the same issue which was reported [here](https://github.com/ondrej1024/crelay/issues/28). Please take a look. There is no proper solution for this at the moment. :disappointed:

Sztoppolino, 2019-07-19:
Hi!
Thank you for your answer. I examined again this issue, and it seems I have the same problem. When I swich ON my relay, crelay can send the command but thereafter (-> try to turn off) my PC lose the connection and crelay crash. I noticed that this problem always occur when I try to switch OFF my relay.

So I tried to confirm [this suspected](https://github.com/ondrej1024/crelay/issues/28) issue. I disconnected the relay module from 230v AC, and I switched on/off the relay for 5 minutes. Crelay didn't crash.

Thereafter, I connected to 230v AC again, and crelay crashed after 30 seconds.

So I agree with you, the 230V can be a good reason. That makes sense.

-------------------------------------------------------------------------------------------------------------------

Did you try ferrit choke? Can I avoid this problem, if I use it?

ondrej1024, 2019-07-19:
> Did you try ferrit choke? Can I avoid this problem, if I use it?

Yes I tried and it helped in my case switching a 24VAC load. However it does not work 100%.

ondrej1024, 2020-11-03:
This is a duplicate of issue #28, therefore closing this one.

closed 2020-11-03 by ondrej1024