ondrej1024/crelay #37: Detection of relay card fails, if there is a usb serial thinghy

browsem, 2020-04-23:
Hi. 
Im doing some stuff on an OpenWrt install.
I have a noname generic usb relay card(HMI), that can be controlled via crelay.
Then i add a usb<->serial ttl thinghy.
Either a cp210x or an ft232 thing.. In both cases crelay latches on to those, and wont recognize the relay any more.
I can still use the usb2Serial to talk to the sensor, but the relay is off line
Is there any way to force is, so it uses the correct card ?

ondrej1024, 2020-04-23:
Try to detect your relay card with the `-i` switch:
`crelay -i`

It will report the serial number of your card. Then you can start crelay specifying that serial with the `-s` switch.

browsem, 2020-04-23:
When i run the -i, i get the following result 
Detected relay card type is HID API compatible relay card (on 0001:0003:00, 8 channels)
i have tried the 0001:0003:00 but that doesn't work

ondrej1024, 2020-04-23:
Can you paste here the exact output of `crelay -i` ? Also the output of `lsusb` would be helpful.

browsem, 2020-04-23:
When i DONT have the CP210x usbSerial plugged in
crelay -i -> Detected relay card type is HID API compatible relay card (on 0001:0003:00, 8 channels)
When i DO have the cp210x thing in it responds with
crelay -i -> Detected relay card type is Conrad USB 4-channel relay card (on Serial number 00                           01, 4 channels)
and the relay card doesn't work
If i use an FT232 thing, same story, but it detects another type of relay card. ( cant test right now)
lsusb ->
Bus 001 Device 005: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP2102/CP2109 UART Bridge Controller [CP2                            10x family]
Bus 001 Device 003: ID 16c0:05df Van Ooijen Technische Informatica HID device except mice, keyboards, and j                            oysticks
Bus 001 Device 004: ID 2341:0001 Arduino SA Uno (CDC ACM)
Bus 001 Device 002: ID 14cd:8601 Super Top 4-Port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

The HID is the relay, the CP2102/CP2109 the serial thing i would like to use.
The arduino, is a workaround thing. Basically i'm using the 16U2 on it as a usb-Serial ( im using it to talk to a sensor)
i added the lsusb -v as a file
[lsusb_v.txt](https://github.com/ondrej1024/crelay/files/4523176/lsusb_v.txt)

ondrej1024, 2020-04-23:
I think this is a very old version of crelay that doesn't support multiple relays and the serial feature. The OpenWRT package has not been updated for a long time.

The current version is 0.14 and I get this when running `crelay -i`
```
Detected relay cards:
  #1    HID API compatible relay card (serial X3AY3)
```

browsem, 2020-04-23:
ahh. That could be it.
I will try and rebuild from source, and see if i can get it working, and who knows. Perhaps even get the openwrt package updated. 
Thx for the help

browsem, 2020-04-23:
Quick update. I did a sysupgrade of the router. This brought me from version 0.10.01 to 0.13 Now i get the following
Detected relay cards:
  #1    Conrad USB 4-channel relay card (serial 0001)
  #2    HID API compatible relay card (serial HW341)
So. Now just one question remains. do i add -s HW341 to the rc.d start file, to get it to use the correct card for the webgui ?

ondrej1024, 2020-04-23:
Unfortunately multiple card support is not yet implemented in the WebGUI. It will always show the first detected relay card. In the latest version 0.14 at least the HTTP API supports multiple cards.

But there is one thing you can do. crelay can be build excluding the relay drivers you don't need. So if you manage to build it for your system excluding the Conrad driver it will not "see"  the USB-Serial converter anymore. Just build with this command:
`make DRV_CONRAD=n`

browsem, 2020-04-23:
That would solve it. 
I think i'm going to turn the demon off for now. I would make more sense for me, to do something a little more custom, and the "dayli" use of the relays, are controlled via a cronjob/AshScripts anyway.
Im using it to control what area my lawnmower robot is going to mow. The sensor im using is an ultrasound one, that i'm interfacing via serial  . the basic operation is, that if the robot has been home for more than 10 minutes, it will flip to the next area.
I have modified my scripts to reflect that it sees 2 cards, and now everything seems to work.
So if i'm going to do anything, it would be to add a couple of buttons to the Lua interface of the router. Something like Area1 Area2 and lock to Area.
But i digress.
Thank you so much for you help

browsem, 2020-04-23:
The problem was solved by upgrading to a newer version.

closed 2020-04-23 by browsem