ondrej1024/crelay #40: Conrad 4 channel card suppoert?

jonaumann, 2020-06-04:
after compiling with "make" and running crelay I got the follwoing message:

crelay, version 0.14

This utility provides a unified way of controlling different types of relay cards.
Supported relay cards:
  - Sainsmart USB-HID 16-channel relay card
  - Generic GPIO relays

why is the conrad card not listed as supported?
Currently is no card attached to the PC.

ondrej1024, 2020-06-04:
Please copy the exact make command you run and its output here. Thanks.

ondrej1024, 2020-06-04:
Does make throw any errors? Probably the conrad driver is not built due to some dependency errors. The detailed printout of the make command would be helpful to find out what is happening.

jonaumann, 2020-06-05:
sorry, forgot to put the make output in the last email.
Here it is, there is just a warning but no error and it says that it
compiles the conrad driver:

*strelen@linux*:*~/dev/crelay/src*$ make clean
[Clean]*strelen@linux*:*~/dev/crelay/src*$ make
[Compile crelay.c]
[Compile relay_drv.c]
[Compile config.c]
[Compile relay_drv_gpio.c]
[Compile relay_drv_conrad.c]
[Compile relay_drv_sainsmart.c]*relay_drv_sainsmart.c:* In function
‘*detect_relay_card_sainsmart_4_8chan*’:*relay_drv_sainsmart.c:197:15:*
*warning: *argument 1 null where non-null expected [*-Wnonnull*]
          if (!*strcmp(serial, (char *)sernum)*)
               *^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~*
In file included from
*relay_drv_sainsmart.c:70:0*:*/usr/include/string.h:136:12:* *note:
*in a call to function ‘*strcmp*’ declared here
 extern int *strcmp* (const char *__s1, const char *__s2)
            *^~~~~~*
[Compile relay_drv_sainsmart16.c]
[Compile relay_drv_hidapi.c]
[Link crelay] with libs -lusb-1.0 -lftdi -lhidapi-libusb -lhidapi-libusb

ondrej1024, 2020-06-05:
Are you sure you are executing the latest crelay binary you have built? After successful `make` comand execute crelay like this in the scr/ directory:
`./crelay`

jonaumann, 2020-06-06:
ahhh, that was the problem. I executed the wrong binary. Sorry for the fuzz and thanks for your help!

closed 2020-06-06 by jonaumann