ondrej1024/crelay #41: API to deal with multiple relays at once

spacelama, 2020-10-29:
process_http_request() (and the comandline, I think) can only deal with one relay at a time per call.  It would be nice if there was a way of dealing with multiple in the one call.

ondrej1024, 2020-11-01:
That's correct. Only one relay at a time is supported for both command line and in each HTTP request.  The command line could possibly be modified quite easily to accept multiple relay parameters. The current REST API instead is not something I will modify any more. I would rather replace it with a new implementation, based on json format. That new API would also support multiple relays.

This has been discussed some time ago [here](https://github.com/ondrej1024/crelay/issues/30#issuecomment-449657001)
You are of course welcome to provide a patch for this :smiley:

spacelama, 2020-11-01:
Missed that bug in my search presumably because it was closed. Yes agree JSON would make more sense.  To answer the motivation question - it would be to better guarantee that all relays have been successfully set.  If the command returns success, all relays have been changed, whereas if you have to call them sequentially, more can go wrong.

I ended up doing them through concatenating several calls to the commandline over a single instance of ssh (with error checking only on the ssh call), but that means sharing ssh keys to all the hosts that need to control the relays.  Keep the bug open and I'll see whether I get enough time to learn JSON libraries :)

ondrej1024, 2020-11-02:
Rather than calling the command line via ssh I would recommend using the REST API. This is the preferred way of remotely controlling `crelay`. At the moment you still have to issue one requests for each relay to control multiple relays, but you don't need to share ssh keys or things like that. The current REST API is rather ugly but I had to implement it this way in order to be compatible with the [PiRelay](https://play.google.com/store/apps/details?id=com.jasonfindlay.pirelaypro) smartphone app. So the new JSON API should be _added_ rather than replacing the old one to not break compatibility with PiRelay. 

A very nice JSON library is [libjansson](https://jansson.readthedocs.io/en/2.13/).

By the way, another way of remotely controlling `crelay` is the MQTT protocol. It's not completely finished but I have already implemented an MQTT client into `crelay` in the [mqtt branch](https://github.com/ondrej1024/crelay/tree/mqtt). Unfortunately, also this lacks support for multiple relay control.