ondrej1024/crelay #43: Stable device path

paragi, 2021-05-30:
Hi Ondrej
Thanks for a handy program.
The path to a device, that has no serial number, is changing depending on other factors than physical connection (linux). This makes it hard to configure for multiple relays, of the same kind.
I need a stable identification to a bunch of relays without serial numbers.

would you accept a pull request, with an alternative path, based on HW bus connection? 
I would much prefer to contribute to an existing project, rather then creating a new one.
I imagine it will be a rewrite of the detection logic,  using the underlying libusb-1.0
I would be unable to test on windows and mac.
Would that be within you scope for this project?

best regards 
Simon Rigét 
https://github.com/paragi/

ondrej1024, 2021-06-02:
Hi Simon,
of course you are welcome to contribute to `crelay` with a pull request.
But before implementing your changes take a look at issue #26. Looks like @derekatkins has implemented a solution for this in libusb.

For testing on Windows or Mac, this is not in the scope of this project. It is meant to run on Linux.

Regards, Ondrej

paragi, 2021-06-02:
Thank you for the reference issue.  That is exactly the issue I'm concerned about.
When there is no serial number, the only way to uniquely identify the device, is via the bus port address. 
I think the solution is to obtaining the bus address via libusb, and using the it to identify the device number, used with the hidapi.
But it requires a rewrite of the detection functionality. 
I will start writing some code now, it you have no other concerns? 

Best regards
Simon

ondrej1024, 2021-06-03:
No concerns, please go ahead with your modifications.