ondrej1024/crelay #49: Interactive Command LIne Documentation

19smuh23, 2024-03-28:
In the Readme file, the interactive command line syntax is given as: crelay [-s <serial number>] -i | [<relay number>] [ON|OFF]. For someone well versed in Linux, the syntax may be obvious from the above. But I am not, so it took me a long time to figure out that you have to repeat the crelay command after the pipe. What worked for me was: crelay -i [-s <serial number>] | crelay  [<relay number>] [ON|OFF]. I also put "-i" before the serial number. Perhaps you could give some example commands in the documentation to make it clear to newbies.

ondrej1024, 2024-04-02:
I agree, the syntax might not be obvious to someone approaching the command line for the first time. I will add some examples in the doc.

Here just a short explanation of the syntax elements:
`<x>` this is a placeholder for a required parameter (you _need_ to provide a value for x)
`[x]` this is an optional parameter (you _can_ provide a value for x if needed)
`x|y` this is an alternative (you need to provide a value _either_ for x _or_ for y)

Brad7173, 2025-05-13:
It took an act of world congress to get this after trying command after command for 2 hours...here it is!!
crelay -s "serialnumber" "relaynumber"

Here's one way to use it in bash:
#!/bin/bash
Relay4StatCheck=$(crelay -s "AQ03CFDT" "4" | awk {'print $4'})
Relay4Status=$(printf "$Relay4StatCheck")
if [ "$Relay4Status" != "on" ]; then crelay -s AQ03CFDT 4 ON
fi
#crelay -s "AQ03CFDT" "4"
#Relay 4 is off
exit