ondrej1024/crelay #50: Problem with serial number (special character)

labmaster, 2024-06-15:
When i start crelay_arm64 with the -i Option to get the serial of the relay card  with
_./crelay_arm64 -i_
the following output is shown:
_Detected relay cards:
  #1    HID API compatible relay card (serial 7▒MBS)_

when i then try to use this serial number to control the relay card with 
_./crelay_arm64 -s 7▒MBS 1 OFF_
it get:
_No compatible device detected._

[image]

Any hint how to deal with this serial number (including such a strange character) ?

ondrej1024, 2024-06-16:
I assume the serial number of your card is corrupted. A correct serial number should contain only printable characters.

However, if you have only one relay card, you can start `crelay` without the `-s` option. It will use the first card which is found.

labmaster, 2024-06-17:
Regarding a corrupter serial number,  i had exactly this System running for over 2 years without any such problem. 
But now i had to reinstall the whole system and after that this problem with the serial number occurred.

Could it be that this has something to do with character coding ?   (i ' am here in germany)
When i copy the string (that one that i get with -i) and paste it into a hex editor, i see that the 'special char' is build form 3 bytes.
So this looks like some kind of UTF-8 coding.

If i remember correctly the wrong character was a 'Q' on the old working system.
'Q' might not available on every charset.

labmaster, 2024-06-17:
if i look at the HID specification for my Relay card i can see that every serial number item is a single byte coded directly inside of one message.
So i guess each such byte that is stored inside of the relay cards would be allowed to be something within the range from 0 to 255
that means even non printable characters could be possible.

ondrej1024, 2024-06-17:
If you need to handle raw bytes as serial number, then you need to change crelay.c in 2 places:

* [Line 935](https://github.com/ondrej1024/crelay/blob/3f1033bf1bd5dd81ca24fcf4c88cb521f7bdc088/src/crelay.c#L935): print the hex presentation of the bytes instead of the string at card detection
* [Line 948](https://github.com/ondrej1024/crelay/blob/3f1033bf1bd5dd81ca24fcf4c88cb521f7bdc088/src/crelay.c#L948): handle serial number parameter as hex byte presentation

I am not sure if also the string compare functions in the specific driver need to be changed.

labmaster, 2024-10-19:
I finally reset the serial number using "ubsrelay", now "crelay" works again as it has done the last 2 years.

When i accessed the card using "crelay" at a bad moment , it seems to be happend during connect/unconnect the relay cards USB,  it got corrupted again :-(

I now investigated in changing the Relayboard firmware (for the Atmel ATMega8) to prevent this happening again. For this i modify the firmware to only change the serial number internally if the change command is received at least 3 times without any other command in between.

We will see if this will prevent this problem in the future.

ondrej1024, 2025-03-04:
This doesn't seem to be an issues with `crelay`. Closing this issue.

closed 2025-03-04 by ondrej1024