ondrej1024/crelay #52: Issue on RPi bullseye and bookworm

seuchato, 2024-10-24:
Hi
I have three pcs: 

1. HP t620, OpenSUSE 15.6, crelay installed as per the description ["Installation from source"](https://github.com/ondrej1024/crelay#installation-from-source) . Works flawlessly with HID based relays
  crelay -i shows
   Detected relay cards:
     #1    HID API compatible relay card (serial 959BI)
Switching it on and off through the webinterface or from commandline works flawlessly

2. A Pi 3B V1.2 with bullseye on it, same installation procedure. 
crelay -i shows the same result as above, but the relais does not react either through webinterface or command line, it simply stays OFF

3. A Pi Zero with bookworm on it, same installation procedure. 
Same problem as with the 3B

What could cause that?
Greez
chris

ondrej1024, 2024-10-29:
Interesting. I haven't tried to use `crelay` on a recent Linux distri. I guess this issue depends on the different versions of the `libhidapi` development libraries that come bundled with the distries. 
It might be necessary to check the changelog of these libraries and find out what are the differences of these versions.

seuchato, 2024-10-29:
Ondrej
It's solved: I ran the latest updates and the problems vanished. The issue can be closed.
Greetings
chris

closed 2024-10-29 by seuchato